jneilliii/OctoPrint-BedLevelingWizard #1: Printer not travelling to next X/Y coordinate

Jmccl89, 2018-07-18:
When I try to move to the next position using the button, with the configuration values left at default, the print head will lift and return to 0.1, but there is no movement to the next x/y position. G Code in terminal is as follows. Octoprint printer profile is setup as a 200x200x180 build area, not sure if that is where the plugin gets that info from.

```
Send: G90
Recv: ok
Send: G1 Z10 F1200
Recv: ok
Send: G1 Xundefined Yundefined F4800
Recv: ok
Send: G1 Z0.1 F1200
Recv: ok
```
Any idea what I've got going wrong here? Thanks.

jneilliii, 2018-07-18:
Can you open you browser's console and run through the wizard and report back what it writes out as you press the button. What version of octoprint are you running?

Jmccl89, 2018-07-18:
Octoprint is at 1.3.8 Stable.
Console outputs this on each button press.

```
packed_core.js?4606961a:14533 Application startup complete
packed_plugins.js?3cdeef58:753 ƒ c(){if(0<arguments.length)return c.tb(c[E],arguments[0])&&(c.ga(),c[E]=arguments[0],c.fa()),this;a.l.oc(c);return c[E]}
packed_plugins.js?3cdeef58:754 Uncaught TypeError: custom_box.x_min is not a function
    at BedlevelingwizardViewModel.self.start_level (packed_plugins.js?3cdeef58:754)
    at BedlevelingwizardViewModel.eval (eval at parseBindingsString (packed_libs.js?eebed41b:10824), <anonymous>:3:76)
    at HTMLButtonElement.<anonymous> (packed_libs.js?eebed41b:10845)
    at HTMLButtonElement.dispatch (packed_libs.js?eebed41b:4738)
    at HTMLButtonElement.elemData.handle (packed_libs.js?eebed41b:4550)
```

RickMcConney, 2018-07-19:
I too tried the plugin and I get the same error as above.

jneilliii, 2018-07-19:
Sorry guys, I'll be looking into this today/tonight. Something got messed up while fixing custom box options.

jneilliii, 2018-07-19:
This issue has been resolved with version 0.1.4.

closed 2018-07-19 by jneilliii