jneilliii/OctoPrint-BedLevelingWizard #4: Only Z Axis Moves

dcs3473, 2018-07-20:
Hello,  I installed your plugin and when I run the wizard only the Z axis moves up and down.  The X and Y stay in the same position.  Is there a setting or adjustment I am missing?

Thanks.

dcs3473, 2018-07-20:
It was installed the day before yesterday.  Not sure on the version.  When I press Next this is what I get.

Send: G90Recv: ok
Send: G1 Z5 F1800
Recv: ok
Send: G1 Xundefined Yundefined F6000
Recv: ok
Send: G1 Z0.1 F1800
Recv: ok

jneilliii, 2018-07-20:
Yeah, that should be fixed in the latest version.

closed 2018-07-20 by jneilliii

dcs3473, 2018-07-20:
That it did.  Thank you very much for the help.  Love the PlugIn.  Little cleaner then the Custom Control Buttons I created to do the same thing.

jneilliii, 2018-07-20:
Awesome, and thanks for the tip.