jneilliii/OctoPrint-BedLevelingWizard #5: would be nice if can be used when touchui plugin is enabled

micdenny, 2018-07-21:
when I run octopi with touchui enabled, I can't see this plugin anywhere, could it be fixed to work in touchui mode?

jneilliii, 2018-07-21:
Realized this is working on a full screen touch screen but not on my phone...reopening.

micdenny, 2018-07-22:
Thank you, this is really useful plugin and now I can use it from the raspberry and the small touch lcd.

micdenny, 2018-07-22:
just one more useless thing, you can add an icon for this tool, now it shows the `?` icon, both on mobile menu and pad menu:

[image]

[image]

micdenny, 2018-07-22:
in the meanwhile check also the routing url, you don't have a routing hashtag for the wizard, so for instance when you change from printer page to the wizard page the url remains `http://octopi.local/#printer` when I expect to see something like `http://octopi.local/#leveling_wizard`