jneilliii/OctoPrint-BedLevelingWizard #12: separate X and Y Axes Offset

Hanks552, 2018-09-10:
im trying to use this BedLeveling with my BL Touch and Klipper... it would be perfect if i could set the X and Y axes offset because of the BLTouch mount offset

jneilliii, 2018-09-10:
If you have a BLTouch I am assuming you are using Auto Bed Leveling, not manual bed leveling.  This plugin doesn't really make sense in that case and is really intended for just moving the nozzle around the bed for manual leveling procedure.  With a future update I plan to allow for custom points to be used, which would accommodate your request, just not sure if this plugin was what you were looking for or not.

Hanks552, 2018-09-10:
i was adjustin the knobs to level the corners, i was using bltouch and the command probe to find the Z height... on marlin auto bed leveling was perfect, but with klipper it just does 1 point...

jneilliii, 2018-10-06:
I've just pushed a new branch with the custom points option.  Enable the option and enter whatever points you want to use for probing in settings.  Install using the url below in plugin manager and report back here so I know if it is working properly prior to official release.

https://github.com/jneilliii/OctoPrint-BedLevelingWizard/archive/0.2.0.zip

jneilliii, 2018-10-10:
Pushed official release 0.2.0.

closed 2018-10-10 by jneilliii