jneilliii/OctoPrint-BedLevelingWizard #13: Popup covers data entry for wizard

GreatBizarro, 2018-09-27:
Is there a way to move where the "instruction of use" popup shows up at, it is covering the input for the wizard on my laptops screen.
Thanks

jneilliii, 2018-09-27:
Do you mean the one that shows when you first install it? Screenshot?

drmulligan, 2018-10-06:
This is a screen size issue.  It does not happen on my browser when full screen, however when I resize by browser and make it narrower I can easily see the overlap reported above.

jneilliii, 2018-10-06:
Oh, I think I know what the OP meant now. If you minimize the other sidebar panels like files it will help in that situation.

GreatBizarro, 2018-11-07:
Sorry for the late response, don't check this mail address often enough.
The one that comes up when you hit start it is at the bottom left of the
screen and covers up the start / stop buttons. I can scroll the octoprint
screen to uncover them but it gives my about 1/4 of the botton between the
popup and the bottom of the screen in wich to press the button.
Thanks

jneilliii, 2018-11-18:
So I've just pushed a new branch 0.2.1 that makes the pop-up "non-blocking" which should allow you to click through the pop up to the buttons. Let me know if that works for you. To test install using the URL below in the plugin manager,

https://github.com/jneilliii/OctoPrint-BedLevelingWizard/archive/0.2.1.zip

PR #18 (2018-12-02): 0.2.1
### Fixes
- Made pop-up instructions non-blocking so if your screen is not 16:9 aspect ratio you can still click the buttons if the pop-up is over the sidebar panel. The pop-up will fade away when you hover the mouse over it like below. Resolves #13.

[image]
closed 2018-12-02 by jneilliii