jneilliii/OctoPrint-BedLevelingWizard #14: More Points

Kassie75, 2018-10-04:
Hi, Thanks for a great tool, it really makes manual bed leveling a breeze.

Can I ask if it is possible to do more than 4 points, in an option box maybe. I have a MK3a 300mm bed with 6 adjustments, and with the current setup the middle front and rear adjustments does not get set..

Thanks

jneilliii, 2018-10-04:
It's on my list to add custom points for the delta guys and for this specific use case. Just not sure when I'll get to it.

Kassie75, 2018-10-04:
Thanks, I didn't even think about the delta guys.
Keep up the good work!!

jneilliii, 2018-10-06:
I've just pushed a new branch with the custom points option.  Enable the option and enter whatever points you want to use for probing in settings.  Install using the url below in plugin manager and report back here so I know if it is working properly prior to official release.

https://github.com/jneilliii/OctoPrint-BedLevelingWizard/archive/0.2.0.zip

jneilliii, 2018-10-06:
I get your point about the heights,  what if they were moved above the list of points? I did consider the request option but couldn't think of a simple way to do it and figured you could just repeat the points in the list if you wanted thi mimic the multiple cycles like the offset method.

Kassie75, 2018-10-08:
Moving the heights outside the bounding box to the top is fine.
I don't know your code structure, but is something like a while loop on the repeat points not an option to repeat the process until the amount of repeats has been reached.

jneilliii, 2018-10-08:
It would take a considerable amount of work to get the repeat option as described.

jneilliii, 2018-10-10:
Pushed official release version 0.2.0.

closed 2018-10-10 by jneilliii