jneilliii/OctoPrint-BedLevelingWizard #15: Settings are editable but don't work until restarting

drmulligan, 2018-10-06:
I was just trying your plugin for the first time and really like it.  Thank you.
However I ran into one issue while tuning the settings.  The Monoprice Select Mini has a metal shroud around the X gantry that gets in the way of the bed leveling screws.  I wanted to test different x/y offset values so I could level the bed using the wizard without moving the hot end or bed.  However when I changed the x/y offset the change would not take effect until I pressed stop then start again.
Please either disable the settings inputs while the wizard is operating or preferably allow the settings to take effect immediately.

jneilliii, 2018-10-06:
That makes sense. Unfortunately I still haven't found a good way to save the settings from the sidebar, so will probably go the route of disabling the inputs while the process is running.

jneilliii, 2018-10-06:
Added the disabled option to version 0.2.0 for testing.

closed 2018-10-06 by jneilliii