jneilliii/OctoPrint-BedLevelingWizard #23: First home X and Y axis...

vodidan, 2019-02-28:
Verry nice plugin, but if you could stop scratching my heatbed it would be nice.

jneilliii, 2019-02-28:
I'm not sure what you mean, the plugin only sends GCODE commands to the printer to move the head. What do your settings look like?

vodidan, 2019-02-28:
I know it only sends GCODE commands,but when I start the bed leveling wizard it always homes the Z axis first and the the X and Y(when I am changing to a longer hotend/nozzle it always scratches the glass heatbed). The settings for the plugin are default.

jneilliii, 2019-02-28:
What firmware does your printer run?  I just double checked and I'm not 100% sure, but the issue may be related to the order in which I'm sending the commands.  When you first start the process it sends

```
G90
GZ{offset_z_travel} F{travel_speed_probe}
G28
GZ{offset_z_travel} F{travel_speed_probe}
```

So maybe just taking out the first Z movement before homing would resolve this issue, or there is some other problem related to absolute positioning (G90 command). Just curious, if you manually home prior to running the wizard does the same issue happen?

vodidan, 2019-03-02:
I am on the move now and cannot check the firmware. The printer is Ender 3 with stock firmware(haven't changed it yet). I do not home it manually prior to running the wizard.

This is my start g-code and I use absolute positioning:

`; Ender 3 Custom Start G-code
G28 ; Home all axes
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G90 ; Set coordinates to absolute
; End of custom start GCode`

jneilliii, 2019-03-24:
I'm thinking of when I get some time to make a new option for "start script" that would be the very first part of the process before probing the points.

alaorath, 2019-08-14:
I just ran into this issue last night. :(

Was changing the nozzle, and may have bumped the Z axis down.  I started the bed leveling wizard without first performing a "normal" auto-home.  The Z move comes first, causing a huge scrape on my pristine glass bed.

Please swap lines 135 and 136 to resolve.

jneilliii, 2019-08-14:
I just incorporated your changes in the 0.2.3 branch.  Try installing the plugin using the url below in plugin manager and let me know if it works as you expected.

https://github.com/jneilliii/OctoPrint-BedLevelingWizard/archive/0.2.3.zip

jneilliii, 2019-08-29:
This new version has been officially released and should resolve this issue.

closed 2019-08-29 by jneilliii