jneilliii/OctoPrint-BedLevelingWizard #29: Z-Axis too low

Jonathan-Garber, 2020-08-06:
I just can't get this plugin to work.
When I click start the z-axis goes down a few mm, then goes all the way up until it hits the limit switch, then drops back down halfway and stops. If I click "Next" the X/Y moves to each point but the bed is halfway down the machine.

If I adjust Z-Height it moves lower down the machine. It seems the axis is inverted but way off.

jneilliii, 2020-08-06:
This sounds like a possible issue with absolute versus relative movement. What firmware are you using on your printer? Can you please share a screenshot of your settings?

Jonathan-Garber, 2020-08-06:
It's a Qidi Tech 1 Dual Extruder
FW: Sailfish 7.8 R01234
I believe this is Replicator clone

Not entirely sure which settings screenshot you need. The Bed Leveling settings are set to their defaults.

jneilliii, 2020-08-20:
What are the commands that you run in a normal terminal window if you wanted to perform a bed probing operation? That's what you want to put into the settings. I don't have any knowledge of Sailfish, and therefore can only possibly help get us to where we need to be for the plugin to work properly.

jneilliii, 2020-08-20:
So the commands that get sent to the printer are determined [here](https://github.com/jneilliii/OctoPrint-BedLevelingWizard/blob/4da341657d8afdbe6a678d1010535fb4ed23e7e0/octoprint_bedlevelingwizard/static/js/BedLevelingWizard.js#L92-L96). It boils down to running the following commands.

```
G91 ; relative movement
G1 Z<z travel height from settings> F<z travel speed settings> ; move nozzle up relatively
G90; absolute movement
G28; home
G1 Z<z travel height from settings> F<z travel speed settings>
```

It then will do a similar approach from that point to the next point, etc. raising the head, moving in xy, lowering head. I wonder if your travel height might need to be negative since it's lowering the bed?

closed 2020-09-12 by stale[bot]