jneilliii/OctoPrint-BedLevelingWizard #30: Physical Button for next Point

Bootscreen, 2020-09-25:
Hello,

is it possible to get an option to connect a physical button on the gpio ports on my raspberry to go to the next point?

jneilliii, 2020-09-25:
Although I like the idea, I personally don't want to get into programming GPIO features into my plugins for a multitude of reasons, one of which is locking them down to Raspberry Pi only OctoPrint instances. I hope you understand.

closed 2020-09-25 by jneilliii

SaKiEQ, 2021-09-21:
Could it be possible to just get a URL POST command that triggers the next position, like an action command. 
Then we can implement this from other sources, such as OctoDash buttons or Enclosure Plugin via shell command etc... 

There are many ways, once one can send the command to advance the level position.