jneilliii/OctoPrint-BedLevelingWizard #33: Test at center improvement

medisoft, 2021-05-06:
May I suggest to add another test point in the center of the bed? To ensure that the center is leveled in addition of the borders.

This makes sense in big printers, where the glass bed can bend a little, and sometimes the borders are a not at the same height as the center.

jneilliii, 2021-05-06:
You can use custom points to achieve this with the current implementation, but will consider as future enhancement.

vintageb3, 2021-05-09:
I understand you can add custom points but can I add a second for this centre test point? Thanks!

jneilliii, 2021-05-09:
The plugin doesn't really care where the points are, it just goes through them sequentially as listed.

vintageb3, 2021-05-09:
Understood. I just thought the option to add a centre point in settings would useful. Thanks for the reply!