jneilliii/OctoPrint-BedLevelingWizard #35: Could it be possible to just get a URL POST command that triggers the next position, like an action command.

jneilliii, 2021-09-22:
Could it be possible to just get a URL POST command that triggers the next position, like an action command. 
Then we can implement this from other sources, such as OctoDash buttons or Enclosure Plugin via shell command etc... 

There are many ways, once one can send the command to advance the level position.

_Originally posted by @SaKiEQ in https://github.com/jneilliii/OctoPrint-BedLevelingWizard/issues/30#issuecomment-924413365_

SaKiEQ, 2021-09-22:
ah I see, so there is no backend on Octoprint to it. Since you manage to do it on browser level, how do you send the 'next position' command to Octoprint ? Is there a way of 'hijacking' that for my idea ?

jneilliii, 2021-09-22:
Not in its current implementation.

SaKiEQ, 2021-09-22:
Philosophically it belongs into OctoDash anyway... If I could figure out how OctoDash is coded, I could contribute, but I haven't figured out its structure and concept. While I manage to make changes and customize functions within Octoprint plugins etc, I am stuck at grasping how OctoDash was coded...

jneilliii, 2021-09-22:
Yeah, angular.js is foreign to me too, the changes that I have PR'd to OctoDash were just copying existing parts already there and adapting for additional plugin support.

SaKiEQ, 2021-09-22:
Yeah, it seems to me you can only understand a angular project if you wrote it your self, or if you invested ages back tracing its workings - which is obviously not fun. I can get the gist of most projects in a short period of time, but this sort of thing is like chinese to me.