jneilliii/OctoPrint-BedLevelingWizard #41: Allow the user to choose the number of passes to run.

jwygralak67, 2023-04-22:
When I run the leveling wizard I typically go around all the points 3 or 4 times.
As it is now, the wizard stops after 2 cycles, and I have to re-start it.

My suggestion is to add a configuration setting to set the number of passes that the wizard runs.
Let it default to 2, then it will be backward compatible to existing user experience.

An alternative suggestion would be to just keep going around the points until the user explicitly stops the wizard.

jneilliii, 2023-04-22:
That was kind of the idea of the custom point options. That will allow you to put the positions as many times as you want.

jwygralak67, 2023-04-22:
Ah. Yes. That would be a manual workaround.