jneilliii/OctoPrint-BedLevelingWizard #42: Level using G30 /BLTouch command with pop-ups and re-test (Enhancement)

tkunchick, 2023-06-17:
would it be possible to add a g30 enhancement to probe the corners one by one, reporting the height at each corner, allowing adjustment and re-probe before moving to next corner?

I was able to do it via gcode and watching the terminal report the height, but no idea how to capture the responses from the g30 command. 

Something to the effect of,

G30 X25 Y200
report and move (First point as baseline, I like the back near the power for the heater. leave that screw alone as much as possible)

G30 X25 Y35
report, retest or accept and move to next point

G30 X190 Y200
report, retest or accept and move to next point

G30 X190 Y35
report, retest or accept and finish

Using Marlin and Octoprint on Rasp Pi 4

Hope I am explaining this well enough.

jneilliii, 2023-06-17:
You might want to check out autobim plugin.

tkunchick, 2023-06-18:
> You might want to check out autobim plugin.

I did, it requires a screen directly attached to the printer, and does not provide location / z height within octoprint.

jneilliii, 2023-06-18:
Ah, I see. It uses M117 messages to instruct you on the direction to adjust, next steps, etc. Could use one of the M117 plugins to have that info displayed within OctoPrint directly as well. I will consider your request and how best to implement though.