jneilliii/OctoPrint-BedLevelingWizard #43: B and C side of printer

AmmaPysz, 2024-02-06:
Is there a way to change only the B and C side of the printer instead of all. I am looking for a offset to go toward the middle of the bed but only on one side of my printer. I do have a CR touch on my Ender 3 V2 Neo. When I use the wizard my nozzle is at the end of the bed on both the B and C side. A and D are correct and don't need to change those.

jneilliii, 2024-02-06:
The only way to really handle that is using the custom points instead of the offset options.

AmmaPysz, 2024-02-06:
Could you please give me an example how I would move it to the left please.

jneilliii, 2024-02-06:
Something like this would be the equivalent of a 200x200 bed with 10mm offset from all sides.

[image]

closed 2024-02-06 by jneilliii